ntrteam/ds_ntrboot_flasher #1: Weird cycle when trying to flash using new 3ds xl usa

rileysrjay, 2017-08-17:
So I tried using this method but with a 3ds (me and some others on gbatemp discussed this and figured it could theoritically work if you took out just the SD card but didn't take out the actual flash cart during the flashing process) but when after I take out the SD card from the flash cart and continue it gets into this weird infinite cycle where it keeps showing different hardware versions and it doesn't stop and I can't actually flash the flash cart. I'm using an r4i gold 3ds rts. Is this a bug within the actual .nds itself or just because I'm trying this with a 3ds which emulates the dsi? Here's the discussion on gbatemp:
http://gbatemp.net/threads/you-can-now-use-an-old-ds-to-flash-ntrboot-flashcarts.481115/page-3#post-7519817

d3m3vilurr, 2017-08-17:
3ds is not a ds and dsl.

rileysrjay, 2017-08-17:
Yes, but someone got it working on a dsi xl on the first page in that thread. So theoritically it would work with the 3ds since it emulates the dsi hardware, right? Read the whole discussion on gbatemp and see if you follow our logic or if we're doing something wrong.

d3m3vilurr, 2017-08-17:
you all are pretty fun.

1. I said just use `DS/DSL`. it means project officially support only these devices, not dsi and 3ds
  - original dsi flasher also uses different command flag for checking chipid.
  - dsi flasher did not use flashcart swap, it makes more risk chance bricks
  - I haven't dsi, and also haven't plan purchases dsi. so I cannot test this devices type & cannot handle all issues.
2. no one succeeds in the thread, but is this buggy? :)

anyway okay, boys. I'll check at tonight on the 3ds, but please do not have hope ;)

rileysrjay, 2017-08-17:
Ok, thanks for clarifying. I personally tried and yes It is pretty buggy on the 3ds with the whole cycle issue and the top screen likes to flash white during the loop. If you want me to I could send a picture of the loop. Also thanks for looking into it, from what you are saying it probably won't happen but it would still be pretty cool if you found a way. :-)

ItsKipz, 2017-08-17:
I just got it to work on an 11.5 o3ds. Why wouldn't it work on a n3dsXL in that case?

d3m3vilurr, 2017-08-17:
@ItsKipz I'm tested with both devices (n3ds/o3ds). not work.

closed 2017-08-17 by d3m3vilurr

d3m3vilurr, 2017-08-17:
I just released new version for supporting dsi.
use this. :)

rileysrjay, 2017-08-17:
Thanks! Will try it out and see how it goes!

Vahnyyz, 2017-08-22:
sorry to open a closed thread, i'm not sure what your policy is on that. I tried using the newest file and following 3dsguide for ntrboot with a NDSL (Nintendo DS Lite) and it kept giving me the cycle that was posted earlier only showing hardware versions over and over. I should note that I'm using an r4 SDHC 3ds, with a 4gb microsd when it loops me in the cycle

Again I apologize if I should not have opened this thread again, was just hoping for an answer

d3m3vilurr, 2017-08-22:
@Vahnyyz not supported R4 SDHC.

Vahnyyz, 2017-08-22:
Thank you sir, I apprecaite your super fast response. I'm sorry to have bugged you >.<!

commit 6d371e31 (2017-08-17): Support NDSI mode

Resolve #1