ntrteam/ds_ntrboot_flasher #2: R4iSDHC.Com FAMILY CARDS

Juanito9001, 2017-08-17:
hey @d3m3vilurr any posibility of this exploit works on the R4i SDHC RTS LITE?
so many people is asking in the kitling space... to do the ntrboot

d3m3vilurr, 2017-08-17:
It related kitling/flashcart_core and need tester for fixing timing issue.

KonzolozZ, 2017-08-17:
r4isdhc.com RTD LITE flashcart details:
ChipID: 000007C2
HW Rev: FFBE0000

Link: http://www.r4isdhc.com/r4-product#a1

If you need tester i'm here.

KonzolozZ, 2017-08-17:
r4isdhc.com RTS LITE flashcart details:
ChipID: 000007C2
HW Rev: FFBE0000

Link: http://www.r4isdhc.com/r4-product#a1

If you need tester i'm here.

d3m3vilurr, 2017-08-17:
I mean, I need test after support flashcart_core :)

Ghioaga, 2017-08-17:
I also can also join the testing, i really need this for a R4i SDHC 3DS RTS ( this website http://r4i-sdhc.com/ ).
If i can help with any info or dumps or anything, you can give me instructions and i will do.
I have an Old3DS, a NDSL. Also I can access NDSIXL, Old3DSXL.

MilkyPL, 2017-08-18:
I can test r4isdhc.com dual core (2017 version).
ChipID: 1D5C137F
HW Rev: 137EA6B5
I have access to Old2DS

Zemogiter, 2017-08-19:
Sorry if I'm asking in a wrong place but does this card work with 3DS Single System guide on 3ds.guide ?
[image]

tajans, 2017-08-19:
@Zemogiter not at the moment

_Sent from my Samsung SM-G930F using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Valeri0p, 2017-08-19:
Guys, please, this repo __is not__ for requesting flashcard support, this is only for the nds mode flasher program.
For that, please go to https://github.com/kitling/flashcart_core/issues/3#issuecomment-323481094 and search if someone already posted an issue for that.
Dstt/sdhc/gold/ace clones are already WIP, so please don't bother the devs if you have one of these.

closed 2017-08-21 by Juanito9001