ntrteam/ds_ntrboot_flasher #3: Some suggestions: flashdump saving and launching from haxxstation

Valeri0p, 2017-08-19:
Hi @d3m3vilurr, I'm here to drop some suggestions for improving this incredible program, maybe you've already thought about it:
  1. Flashdump backup load and save from memory; since the most common method to install ntrboot will be directly from 3ds, a lot of people will need a backup of our flashcard to restore its functionality, and since there will be more and more supported flashcards, It'll be harder to provide a backup online for every card; Introducing an option to save the dump can solve this problem, on flashcard SD (if possible) or on network (if it's not too hard to implement an ftp server to save/load the dump on PC, for example); this last option connects with the second suggestion:
  2. Made the application executable via Haxxstation; haxxstation is an exploit that permits to load ntr mode homebrew, sent from an hacked *DS or Wii, via download Ds mode, without any need to have homebrew access on the destination console; it can help some edges cases, were you have a compatible card but no way to flash it directly; I know that the flashrom is placed inside the executable for now, but if we can load it like I've said before, It won't be a problem; anyways, I understand if this is a too arduous work.

Lastly, I want to report this fork from @saibotu that implements most of the first suggestion, and refactor most of the program graphics. I don't know why there isn't any pull request yet ¯\_(ツ)_/¯
https://github.com/saibotu/nds_ntrboot_flasher

I hope that these suggestions can help with the program future; thanks again to everyone ;)

d3m3vilurr, 2017-08-20:
1. file save requires libfat and dldi. and flashcarts have [different dldi][ref0]. real reason is, if you can find flashcart backup binaries, you don't need backup process.
2. it can't help ntrboothax process.
3. I know she's work, and probably not finished.
  but this method also has the same problem, ak2i does not need restore action, and other carts lost nds flashcart feature. so we should use ntrboot_flasher for the restore.

[ref0]: https://www.chishm.com/DLDI/

d3m3vilurr, 2017-08-20:
Ah I was misunderstand, you mean restore using haxxstation?

closed 2017-08-20 by d3m3vilurr

Valeri0p, 2017-08-20:
1. I've forgotten about dldi, but iirc most modern flashcard should auto dldi patch the nds when loading it, or provide a manual patcher; if I'm right, it should be a good comprise if you don't need to remove the flashcard/SD card ;) (ex:flashing on 3ds/dsi)
  1.5 load a flashrom from/save a flashrom to network storage should be platform independent, but maybe it's a bit hard to implement/less reliable when transferring data; it may need some checks to verify if the transfer gone properly.

About the backup availability, I think that when more flashcards will be added to flashcore, a lot of different flashdumps will popup from similar clones, and it will be very hard to provide a backup for everyone; since this will be the most common method of installation, I think that a backup function can be really useful. Also, I've noticed that for some cards the backup will be >=2 Mbytes, and maybe the ds mode ram isn't enough to store that, even in hotswap mode...

  2. You can send any payload with haxxstation, and use all the functions of the flasher(access the card slot,do installation/restore), the only limitations are the dimensions of the payload (2/3 Mb maybe?) since it'll be stored on the ds ram, and the inability to reach the flashcard SD slot directly. A combination of haxxstation+save flashrom backup to network+load ntrboot_flashrom from network can bypass this issues. IMHO this method has a great potential, but maybe It's a bit complicated to adapt the application for that.

About restoring, I also think that is better to restore ntrboot_flasher on 3ds, but only If you have a backup ;) but doing that from here won't be a bad addition 

Sorry for the wall of text, I want to clarify this discussion; I hope that these ideas can help the development of this tool. :)

d3m3vilurr, 2017-08-20:
1. actually your opinion is generic and if this project only supports ak2i, I also try that.
  but sadly, this project supported multiple carts, it makes the risk for supporting multiple carts in the future. I don't want increase built binaries. 
2. payload limit will lead more complex logics, probably this feature has less benefit than working cost

Valeri0p, 2017-08-20:
Sorry, I don't understand 1, do you mean there won't be any way to at least backup the original flashrom, any were? :(
And Ok about 2, I known It isn't the best environment.

d3m3vilurr, 2017-08-20:
I mean didl can leads multiple building for target flashcarts. I don't want this.