ntrteam/ds_ntrboot_flasher #6: Chip Id and HW Rev changing

NicknameGoesHere, 2017-08-24:
Every time I open the flasher, it changes the Chip Id and Hw Rev

MilkyPL, 2017-08-25:
its an issue with the flashcart or flashcart_core, some carts do this and there is no solution yet

_Sent from my Htc HTC Desire 820 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

d3m3vilurr, 2017-08-25:
@NicknameGoesHere what flashcart?

Zemogiter, 2017-08-25:
Same issue here with R4i 2017 Dual-Core RTS.
http://www.r4isdhc.com/wp-content/themes/nds9/images/w/31.jpg

[image]

[image]

Valeri0p, 2017-08-26:
I've also seen it with my flashcard, it happens (on ntr boot flasher) when you don't insert the micro SD.

Zemogiter, 2017-08-26:
@Valeri0p I have my sd card inserted though.