ntrteam/ds_ntrboot_flasher #8: DSi Flasher version

anuserlol, 2017-08-29:
I know that the DSi version doesn't have a restore flash option,but why?
Also, it's posible to use the NDS version on a DSi?

Valeri0p, 2017-08-30:
Restore flash needs a memory to save the flash backup.
Since this versions can't (for now) save on SD card, the only medium is the nds ram, but since it's erased when you shutdown, you need another device to keep turned on when you install sighax,and since you can do card swap only on Ds lite/fay, these are the only devices viable for that.

d3m3vilurr, 2017-08-31:
Sadly, DSi mode can't swap cartridge. It means, if I add restore feature, you cannot use cartridge for the ntrboot exploit. So restore is useless on dsi/3ds

closed 2017-08-31 by d3m3vilurr