ntrteam/ds_ntrboot_flasher #19: R4i 3DS Gold Pro

Nyvia, 2017-10-15:
This flashcart doesn't seem to work with the flasher. It's the 2017 model that's the same hardware as basically every other cart on r4isdhc.com. it's been confirmed as working with the flashcart-flasher from kitling but doesn't work here, according to people with the same cart as I.

bcecchinato, 2017-10-15:
Tested as well on this card : 
[image]

When running the flasher, the error "Flashcart setup failed" is displayed. I've tried to rebuild a flasher version with the DEBUG flag, but no other logs are prompted.

Regards,

Nyvia, 2017-10-15:
Yep that's the cart I'm using (sans the crooked label).

Endmost, 2017-10-15:
I have the same problem using the following card:
[image]

Marioddd1, 2017-10-15:
I had the same issue, however with the 2016 version of the same cart. 
[image]

pbanj, 2017-10-15:
add me to the list. i have a 2014. the ntrboot flasher works but not this

Nyvia, 2017-10-17:
Confirmed as still not working on 3.3.

jechus, 2017-10-17:
@Marioddd1 same here

Radtendo, 2017-10-18:
Hopefully it's added as a compatible cart in a later update. Until then we'l just have to wait. Which for CFW is very worth it.

Nyvia, 2017-10-18:
It seems that it does work on DS/DS Lites though. It's an issue with DSis

Radtendo, 2017-10-19:
Sadly no, this specific cart that me and OP have don't work. I've tried it on my DSLite several times. Someone else even confirmed it isn't supported yet. 
There's a way to get the hax to recognize the cart but it isn't worth it seeing as it usually just gets stuck trying to flash onto an unsupported cart.

Nyvia, 2017-10-21:
Confirmed as working on this card now

closed 2017-10-21 by Nyvia