ntrteam/ds_ntrboot_flasher #21: R4i 1.4.1 Upgrade (R4i-SDHC.com) says "flashcart setup failed"

patrizl001, 2017-10-15:
^
it should work with the newest update since it's an R4i-SDHC card, but it isn't.

[image]

tomsterley, 2017-10-21:
Try the experimental flasher
https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/issues/23#issuecomment-338309340

patrizl001, 2017-10-21:
I used it. It allows me to install, but after it shows the percentage (which was at 0) the bottom screen just goes black after a moment and doesn't display anything. Don't know if that means it worked or not.

Hoshimon, 2017-10-21:
I have the same card, with one difference, it's .hk instead of .com
Same problem with the experimental flasher.

figgyc, 2017-10-28:
.hk card won't work it isn't supported
anyway try the new release of ds_ntrboot_flasher

patrizl001, 2017-10-28:
new release gives the exact same result. Gets to the install screen and then the bottom just goes black.

maza2, 2017-11-18:
I have the same card but I was apparently able to go through the whole flashing process which took about a minute to two.
But after inserting it into my 2DS and performing the key actions it didn't work.

It seems to be flashing something to the card. Maybe some offset is incorrect or something like that.
How can I provide further information about the card to solve the issue?
I have an NDS Lite a stock DSi and a stock 2DS.