ntrteam/ds_ntrboot_flasher #23: R4i 3ds RTS (r4i-sdhc.com)

kuwoh, 2017-10-15:
says flashcart setup failed (using dsi xl if that gives any more info)

adbash12, 2017-10-15:
Same i really need help

cachepollution, 2017-10-15:
Same here. Tried with 2 different flashcarts (both R4i-SDHC 3DS RTS - board B6-V2.7) on a stock N3DS XL and an original NDS. Both the DS and DSi versions said _flashcart setup failed_. After inserting and removing the flashcarts at different times, I got the flasher to work on the NDS (it flat out refused to recognise the cart on the 3DS) but it always gets stuck at 0% while writing NOR.

Here are the steps to get to the writing NOR part:
1. Boot up the DS mode flasher.
2. Highlight the _R4iSDHC_ entry in the menu. __Don't press A yet__.
3. Take out the flashcart, remove the microSD card and put the cart back in the system.
4. A to continue.
5. A to inject ntrboothax.
6. The next screen says _flashcart press \<A\>_. __Don't press A yet__.
7. Eject the flashcart once again, put it back in and then press A. 
8. A again to inject ntrboothax.
9. A for _RETAIL_ unit (in my case).
10. The flasher gets stuck Writing NOR at 0%.

- If I turn off the DS at step 10 nothing gets written to the cart. It still shows up as Touch! Bomberman Land and boots into the r4i kernel fine.
- I tried using one flashcart to get to step 6 and then swapped it out for my 2nd one to see if it was expecting a different cart than the one used to boot it up. Still didn't work.
- I also tried leaving the console on for 30'+ but it never went beyond 0%.

Parpalloto, 2017-10-15:
Exactly what I did on mine and it didn't work either.

grilledpineapple, 2017-10-15:
Yup flashcart setup failed on a 3DS.

knowbeard, 2017-10-15:
Tried 2 RTS lite flashcards on 2 DS lites and had the exact same results as you.

WereWaffle, 2017-10-15:
tested my r4i 3ds rts card with ak2i_ntrcardhax_flasher dsi .nds. Experienced the same problem as OP.

kuwoh, 2017-10-16:
@cachepollution followed your steps still gives me flashcart failed, tried on my 2ds and dsi xl

cachepollution, 2017-10-16:
@kuwoh The steps only seem to work on an _original_ DS. I couldn't get it to work on my 3DS no matter what. Maybe it has something to do with the firmware on newer consoles?

kuwoh, 2017-10-16:
sadly i do not have access for original ds consoles

cachepollution, 2017-10-16:
Did you try the DS flasher? The DSi one seems to be completely broken. I guess we'll have to wait until the issue gets fixed. Besides, even if you did manage to get to step 10, it still wouldn't flash anything to the cart. Perhaps someone with an original DS or a DS Lite could test the steps to see if they get a different result?

kuwoh, 2017-10-16:
yes i've tried the ds flasher multiple times on both of my systems, sadly i do not know anyone with an original ds or ds lite console

ocedalv, 2017-10-16:
I was able to reproduce the same issue, but my cart is a R4 SDHC Gold Pro 2016 (from r4sdhc.com). I am able to flash and restore using NTRBOOT_FLASHER, but not using AK2I.

MilkyPL, 2017-10-16:
I have the same problem on r4isdhc.com dual core rts on old 2ds

KonzolozZ, 2017-10-16:
Problem is same with RTS Lite 2017 on DS and DS Lite.

Parpalloto, 2017-10-16:
I have tried it on a R4i SDHC RTS Lite on an original NDS, a NDSI and a N3DS and it didn´t work.

Shymain, 2017-10-17:
Can confirm same problem persists with R4i SDHC Dual-Core RTS from r4isdhc.com

d3m3vilurr, 2017-10-17:
can you retest with latest release?

demonfurry, 2017-10-17:
having same issue as OP using n3ds xl, r4i-sdhc rts same board revision even with the latest release

kuwoh, 2017-10-17:
^^

Rikimaruaxu, 2017-10-17:
Same here. R4I SDHC dual core RTS freezes when you select the flashcart type on a Nintendo DS lite.

cachepollution, 2017-10-17:
Nope. Still no luck with v3.3. The DSi flasher fails to recognise the cart on N3DS and the DS flasher crashes upon launch, both on the NDS and the N3DS.

ocedalv, 2017-10-17:
Similar result for me as @cachepollution 

N3DS with DSI flasher fails to recognize the cart.
NDS with regular (non-DSi) flasher launches but freezes when I eject the cart / microSD as instructed.

ocedalv, 2017-10-17:
Similar result to @cachepollution 

N3DS with DSi flasher fails to recognize the cart.
NDS with regular (non-DSi) flasher launches but freezes after I select the family of carts, when I eject the cart / microSD as instructed.

knowbeard, 2017-10-17:
Tried 2 different "2017 RTS lite r4isdhc.com B6 V2.8" cards with the nds flasher (non dsi) on a ds lite and I get the same results as @ocedalv and @cachepollution.

grilledpineapple, 2017-10-17:
Tried the newest release on my R4i-SDHC 3DS RTS and O3DS. Still getting Flashcart setup failed for both the NDS and DSi version. Even tried ejecting and removing the SD card.

yoshigamer478, 2017-10-17:
i get issue well trying to flash the R4i 3DS RTS on the latest relase of the ak2i_ntrcardhax_flasher using my orignal nintendo Ds heres the problem the program on the ds loads up fine but when you eject the flash cart out of the DS and take out the Micro SD card and put the Flash cart back into the Ds and hit A on the R4I SDHC family option it just Freezes and dose nothing and i tryed it many time and no luck i even tried the DSI ak2i_ntrcardhax_flasher on my 3ds but it just says flash cart fails setup

d3m3vilurr, 2017-10-18:
actually, 3.3 version cannot support r4i sdhc. it occur [last commit](https://github.com/kitling/flashcart_core/commit/46f650b00d563b354b5c4596352f624a66e605a0) of flashcart_core.
this commit will support old r4i sdhc, but now requires flashcart reset (not softly)
problem is, I don't know right ds steps like 3ds's flashcart reset steps.
I need more time. sorry.

cachepollution, 2017-10-18:
Does this mean that the last commit to flashcart_core broke the flasher or did it actually add support for r4i-sdhc _for the first time_ on the NDS? In other words, could releases prior to v3.3 theoretically support ntrboot injection on these flashcarts? Or is it only possible after the last commit?

MilkyPL, 2017-10-18:
How can i reset the flashcart using 3ds?

d3m3vilurr, 2017-10-18:
@cachepollution 
1. last commit only broke nds version & actually prev commit can support r4i sdhc (but partially).
2. so theoretically I can release with prev commit but I'll try to fix with current commit until I gave up.

@MilkyPL [here](https://github.com/kitling/ntrboot_flasher/blob/ce47ce0/source/platform_ntrcard.cpp#L69-L105)

Shymain, 2017-10-18:
Problem still exists on my end with v3.3, R4i SDHC Dual-Core RTS 2016 for 3DS from r4isdhc.com.

cachepollution, 2017-10-18:
@d3m3vilurr I see. Thanks for clearing that up. Guess it's best to stick to the newer commits then.

Virgilius1995, 2017-10-19:
Same issue with http://r4i-sdhc.com/
Any alternative software?

agescospreader, 2017-10-19:
i have same issue with r4i rts 3ds r4i-sdhc.com   flashcart setup failed

d3m3vilurr, 2017-10-20:
if someone have a time, please test this & copy log messages
[ak2i_ntrcardhax_flasher_dsi.nds.zip](https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1402932/ak2i_ntrcardhax_flasher_dsi.nds.zip)

tomsterley, 2017-10-20:
R4 SDHC RTS GOLD PRO (with sd inserted, tested on dsi with dsi flasher)

flags: 5406020 1 0 1
Sending cmd: 99 01 04 00 00 00 00 00
flags: 7586000  1 0 1
step 0: c074666

Flashcart setup failed

ocedalv, 2017-10-20:
I have a R4i Gold Pro 2016 (r4isdhc.com)
Here we go:

flags 5406020 1 0 1

Sending cmd: 99  01  04  00 00 00 00 00

flags 7586000 1 0 1

setp 0: 92224D68

flashcart setup failed

d3m3vilurr, 2017-10-20:
please retest with this.

[ak2i_ntrcardhax_flasher_dsi.nds.zip](https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1402987/ak2i_ntrcardhax_flasher_dsi.nds.zip)

tomsterley, 2017-10-20:
@d3m3vilurr 
R4i SDHC RTS GOLD PRO 2017 [r4isdhc.com](with sd inserted, tested on dsi with dsi flasher)

Retest with new file

flags: 5406020 1 0 1
Sending cmd: 99 01 04 00 00 00 00 00
flags: 7586000 1 0 1
Step 0: dee7a45f

Flashcart setup failed

cachepollution, 2017-10-20:
Tested with R4i-SDHC 3DS RTS on N3DS.

`flags 5406020 1 0 1`
`Sending cmd: 99 01 04 00 00 00 00 00`
`flags 7586000 1 0 1`
`step 0: B72DA6BB`
`Flashcart setup failed`

d3m3vilurr, 2017-10-20:
here please retest.
[ak2i_ntrcardhax_flasher_dsi.nds.zip](https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1403014/ak2i_ntrcardhax_flasher_dsi.nds.zip)

ocedalv, 2017-10-20:
I have a R4i Gold Pro 2016 (r4isdhc.com)

flags 5406020 1 0 1
sending cmd: 99 01 04 00 00 00 00 00
flags 7586000 1 0 1
step 0: 5d06f072

Flashcart setup failed

d3m3vilurr, 2017-10-20:
@ocedalv  you looks not last one.

tomsterley, 2017-10-20:
@d3m3vilurr 
R4i SDHC RTS GOLD PRO 2017 [r4isdhc.com](with sd inserted, tested on dsi with dsi flasher)

Retest with new file 2

flags: 5406020 1 0 1
Sending cmd: 99 01 04 00 00 00 00 00
flags: 7586000 1 0 1
step 0: b3105a3e
Sending cmd: 68 00 00 00 00 00 00 00
flags: 75860000 1 0 1
Sending cmd: 99 01 04 00 00 00 00 00
flags: 7180000 0 0 1
step 1: 0 (Press A)
FLASHCART INIRIALIZE FINISHED
PRESS A
<dsi shutsdown>

d3m3vilurr, 2017-10-20:
@tomsterley that's totally what I need. I made to kill at that time. because I'm not finished yet :)

cachepollution, 2017-10-20:
Retest results:

`flags 5406020 1 0 1`
`Sending cmd: 99 01 04 00 00 00 00 00`
`flags 7586000 1 0 1`
`step 0: 892CF327`
`Sending cmd: 68 00 00 00 00 00 00 00`
`flags 7586000 1 0 1`
`Sending cmd: 99 01 04 00 00 00 00 00`
`flags 7180000 0 0 1`
`step 1: 0`
`FLASHCART INITIALIZE FINISHED!!!`

And then the flasher quits when I press A.

tomsterley, 2017-10-20:
@d3m3vilurr Ah ok, I didnt know if the dsi shutting down was ok or not

ocedalv, 2017-10-20:
I have a R4i Gold Pro 2016 (r4isdhc.com), using O3DS @ 11.6 (U)

flags 5406020 1 0 1
sending cmd: 99 01 04 00 00 00 00 00
flags 7586000 1 0 1
step 0: 192fe428
sending cmd: 68 00 00 00 00 00 00 00
flags 7586000 1 0 1
sending cmd: 99 01 04 00 00 00 00 00
flags 7180000 0 0 1
step 1: 0
FLASHCART INITIALIZE FINISHED!!!

Flasher quits when A is pressed.

agescospreader, 2017-10-20:
R4i-sdhc 3ds rts (r4i-sdhc.com)
flags 5406020 1 0 1
press A
sending cmd: 99 01 04 00 00 00 00 00
press A
flags 7586000 1 0 1
press A
step 0: FCE2F045
press A
sending cmd: 68 00 00 00 00 00 00 00
press A
flags 7586000 1 0 1
press A
sending cmd: 99 01 04 00 00 00 00 00
press A
flags 7180000 0 0 1
press A
step 1: 0
press A
flashcart initialize finished !!!
press A
software closed

tomsterley, 2017-10-20:
@d3m3vilurr Any idea when a full release will be ready?

trbocode, 2017-10-20:
He uploaded this on discord, try this. Please test only if you have a hacked 3DS.
[ak2i_ntrcardhax_flasher_dsi.zip](https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1403142/ak2i_ntrcardhax_flasher_dsi.zip)

tomsterley, 2017-10-20:
@trbocode This actually worked!

ocedalv, 2017-10-20:
@trbocode worked for me too!

agescospreader, 2017-10-20:
worked for me to

zacharypsp, 2017-10-20:
What would happen if you didn't have a hacked 3 ds and flashed it trbocode

zacharypsp, 2017-10-20:
It seems to have flashed properly but i cant seem to get it to boot in gm9

ocedalv, 2017-10-20:
@zacharypsp you have to rename it to boot.firm and put it in the root of the SD card. but this thread is not for this kind of support, please use the discord channel:

https://discord.gg/C29hYvh

demonfurry, 2017-10-20:
the link from trbocode worked great on my bone stock new3dsxl and the r4i-sdhc RTS card currently finishing installing everything

grilledpineapple, 2017-10-20:
@demonfurry Same thing here. Just used my R4i-SDHC 3DS RTS on a stock o3DS. Thanks a lot to every one who helped bringing this to life.

yoshigamer478, 2017-10-20:
@demonfurry @grilledpineapple so it worked for you guys you where able to flash the R4I SDHC 3DS RTS fine so should it be ok for me to do it on my stock 3ds

[19 more replies]

closed 2017-10-27 by d3m3vilurr