ntrteam/ds_ntrboot_flasher #25: R4isdch dual core setup failed

rjose2002, 2017-10-15:
have a r4i dual core that worked with ntrboot_flasher and doesn't work with the dsi one

benji200, 2017-10-16:
i have r4 dual core 2013 with the page in www.r4isdhc.hk and dont work with ntrboot flasher and dsi , is this card, sorry for my bad english, idont speak english, im a mexicano, o sea que hablo español
[image]

benji200, 2017-10-16:
r4 dual core 2013 
windbond
25Q16BVSIG1236
-----------
PD910B
1304
-----------
203K
KR465

rjose2002, 2017-10-16:
Needs to say R4isdhc.con front and back, other doesn't work at least with flasher
[image]

d3m3vilurr, 2017-10-17:
please retest with latest release.

SohaibArif, 2017-10-17:
It's not working on 2016 Variant tried both the NDS as well as DSi file.

SohaibArif, 2017-10-17:
I am using v3.9 firmware on the flashcart .The website mentioned on the card is www.r4isdhc.com

SohaibArif, 2017-10-17:
[image]

yitzh, 2017-10-17:
apparently it can only be with ntrboot_flaser for now

jason0597, 2017-10-21:
Try the experimental flasher, it should work
(Note: Proceed at your own risk!)

https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1403142/ak2i_ntrcardhax_flasher_dsi.zip

zarzarzarzar, 2017-10-23:
The link that @jason0597 gave worked for me and I installed the cfw without problems

Virgilius1995, 2017-10-26:
Hey thanx it worked(-;

d3m3vilurr, 2017-10-27:
okay, comments don't match the first report.

closed 2017-10-27 by d3m3vilurr