ntrteam/ds_ntrboot_flasher #26: R4iSDHC Gold Pro support?

Radtendo, 2017-10-17:
Hi, I have an R4i SDHC.
It's an R4i Gold Pro from what I can tell on the official r4isdhc website. Mine came without a label but I have ensured that it works. I was wondering if support for this card will be added in the future, as it seems it does not work with it. 
I always get a "flashcart setup failed" error when attempting to use both the nds and dsi files. I am awae that the DSi file does not have r4isdhc as a option (at least the one I got from 3ds.guide, anyway) and I'm just curious if it will be at some point. 

Thanks, Radtendo

jechus, 2017-10-17:
Same happened to me, I have a R4i SDHC Gold Pro 2016, and when I tried to flash it, I just got a "flashcart setup failed" error message. I already checked, and the flashcart I own is indeed original, so I got confused why it wouldn't work, since theorically it's already supported

Radtendo, 2017-10-17:
[image]
if it helps, this is what I got.

cachepollution, 2017-10-17:
Have you tried it with the latest release (v3.3) of the flasher?

Radtendo, 2017-10-17:
Yes I have, that's the only one I've been using, in fact.

cachepollution, 2017-10-17:
The flasher is a bit wonky atm. There is a way to maybe get it to recognise your cart (#23) but it gets stuck trying to inject ntrboot anyway, plus it doesn't work on v3.3. I'd say hold off for a while until the flasher gets fixed, especially if you don't have a backup cart.

Radtendo, 2017-10-17:
Yeah, I've become content with just waiting until an update or something that adds support for this cart. Which seems likely, seeing how I'm not the only one who's posted an issue related to this cart.

jason0597, 2017-10-21:
Try the experimental flasher, it should work
(Note: Proceed at your own risk!)

https://github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/files/1403142/ak2i_ntrcardhax_flasher_dsi.zip

fontainl, 2017-10-21:
it seems to work with R4i Gold Pro, thank you very much!

fontainl, 2017-10-21:
[image]

fontainl, 2017-10-22:
Have you got a tips for restore the card after the procedure please?

fontainl, 2017-10-22:
Do you have a tip to restore the card to its original state? Thank you in advance.

fontainl, 2017-10-22:
I resolved myself. Check the correct date of your console and then inject "r4i_sdhc_dual_core_2014_nor"backup following instructions:
And now all is ok! Enjoy!

Now everything is ok. have fun!

Instructions
Power off your device
Insert your SD card into your computer
Create a folder named ntrboot on the root of your SD card
Copy backup.bin from your flashrom backup .zip to the /ntrboot/ folder on the root of your SD card
Create a folder named payloads in the luma folder on your SD card
Copy ntrboot_flasher.firm from the ntrboot_flasher .zip to the /luma/payloads folder on the source 3DS’s SD card
Reinsert your SD card back into your device
Insert your ntrboot compatible DS / DSi flashcart into your device
Launch ntrboot_flasher by holding (Start) during boot
Read the red screen warning
Press (A) to continue
Select your flashcart
If you do not see your flashcart in the list at the top, read the bottom screen for more info on each option
Select “Restore Flash”
Press (A) to proceed
Wait until the process is completed
Press (A) to return to the main menu
Select “Exit” to power off your device

fontainl, 2017-10-22:
I resolved myself. Check the correct date of your system and then inject "r4i_sdhc_dual_core_2014_nor" backup.bin to your R4 SDHC Gold Pro, following instructions:

Now everything is ok. have fun!

Instructions
Power off your device
Insert your SD card into your computer
Create a folder named ntrboot on the root of your SD card
Copy backup.bin from your flashrom backup .zip to the /ntrboot/ folder on the root of your SD card
Create a folder named payloads in the luma folder on your SD card
Copy ntrboot_flasher.firm from the ntrboot_flasher .zip to the /luma/payloads folder on the source 3DS’s SD card
Reinsert your SD card back into your device
Insert your ntrboot compatible DS / DSi flashcart into your device
Launch ntrboot_flasher by holding (Start) during boot
Read the red screen warning
Press (A) to continue
Select your flashcart
If you do not see your flashcart in the list at the top, read the bottom screen for more info on each option
Select “Restore Flash”
Press (A) to proceed
Wait until the process is completed
Press (A) to return to the main menu
Select “Exit” to power off your device

Radtendo, 2017-10-22:
Success! I've now got CFW on my N3DSXL with this cart! Thanks Jason!

webodan, 2017-10-23:
Nice, i could flash ntrboothax with my r4isdhc gold pro 3ds rts 2017 using the flasher that @jason0597 posted. I suggest merging whatever changes this version has regarding this card's support, because it just werks.

Thank you very much!

gamegearx, 2017-10-24:
So happy I found this thread.  I am looking at getting CFW on my New 3DS XL, but I want to make sure I buy the right card.  I am looking at buying this card: [https://www.r43ds.org/products/R4i-Gold-Pro.html](https://www.r43ds.org/products/R4i-Gold-Pro.html) and it looks like the one in the picture and is even the 2017 version and I guess comes with a label to show it.  I just want to make sure I get the right card as this one seems the cheapest considering it is two for the price of one right now so I figure that way if I can't flash it back to the original state I can have hacked and the other one for ds games.  Just don't want to waste money if it's not the right one.

fontainl, 2017-10-24:
This card works well and it is possible to restore it to the original state but it has a timebomb!
I advise you "R4I GOLD 3DS(RTS)" (r4ids.cn)  that does not have timebomb ans is 100% compatible with ntrboothax.
[image]
[image]

closed 2017-10-27 by d3m3vilurr