ntrteam/ds_ntrboot_flasher #27: Instructions needed to compile ak2i_ntrcardhax_flasher from source

viraniac, 2017-10-17:
I am trying to compile the ak2i_ntrcardhax_flasher in debug mode

Copy 5 binaries

I found 2 in ntrboot_flasher's data directory, 2 from bootstrap9-hidden project, from where am I supposed to copy blowfish-ntr.bin file?

Milanpsmp, 2017-10-17:
https://mega.nz/#!lTogUZpT!y34yPExuEVE7w9byFrBVBvCcpDtqZoL62XCkDsKZIOo

viraniac, 2017-10-17:
Thanks for sharing. But where did it came from?

Milanpsmp, 2017-10-17:
I don't know actually, a developer just gave it to a friend of mine and he gave it to me

cachepollution, 2017-10-17:
The link is originally from a [commit](https://github.com/figgyc/ak2i_ntrcardhax_flasher/commit/24799b756f285549e167d2231ccebc2e21eaa161) to a fork of this project. I don't know where it came from though.

d3m3vilurr, 2017-10-18:
https://github.com/d3m3vilurr/ak2ifirm/commit/3fedaefc788204f6459cc73f3f40bfda1640e21f you can use ak2ifirm's feature.

closed 2017-10-18 by d3m3vilurr