ntrteam/ds_ntrboot_flasher #33: Got R4iGold 3DS (r4igold.cc) (that should be a SDHC clone) not working.

Hoshimon, 2017-10-20:
As the title says, I got a flashcard that other people has described and found to be an SDHC clone, and it's not working, here's some photos of the outside of the flashcard.
If it's proven to be interesting enough, I can open it to take photos of the inside of the cart, and I can provide whatever info's possible to get with just a normal 3DS (magnethax might be first chance I get to actually hack since I got interested on the matter).

The actual problem when running the flasher is that all setups say "Flashcart setup failed press <A>".

[image]
[image]

Thanks beforehand for any help I could get or could give.
Loving how big magnethax has been for a lot of people! <3

Hoshimon, 2017-10-20:
I saw over at //github.com/d3m3vilurr/ak2i_ntrcardhax_flasher/issues/23# an upload from trbocode that actually "starts" flashing, but after a second or two the bottom screen goes black and nothing happens until you restart, maybe that's helpful in some way?

jason0597, 2017-10-21:
Does ntrboot_flasher support this flashcart? From what I recall, the only working flasher is that modified chinese version on gbatemp that cannot be reverted

trbocode, 2017-10-21:
@Hoshimon it doesn't remotely look like an r4isdhc clone
This the R4isdhc https://user-images.githubusercontent.com/8488758/31848053-2206923e-b63a-11e7-8485-c51954582cb2.png

Hoshimon, 2017-10-21:
It does look really differently, I went with it being a SDHC clone since I saw a lot of forums with these r4igold.cc with years engraved into them that went as SDHC clones.
Also the SDHC menu in ntrboot flasher is the only one that goes throught.
Guess It's just out of compatibility then.
I know people have already too much on their hands with this, but is there any info I can give so this could have support in the future?
I don't know if this cheap flashcard might be interesting to provide support too, I only know these sell commonly in Latin America for like 10 bucks (at least in Chile).

Thanks for the answer!

Hoshimon, 2017-10-21:
@jason0597 It's supposed to work, but it actually doesn't flash anything at all, just reboots.
This made me think that either the card is a clone of anything but the r4igold.cc or that the flashcard itself is unflashable.
Thanks for the answer!

d3m3vilurr, 2017-10-22:
this chip isn't r4i sdhc. current flashcart_core does not support this.

jason0597, 2017-10-23:
Refer to this post:

https://gbatemp.net/threads/release-ntrboot-for-r4igoldcc-ards-ez-ardsi.483802/

d3m3vilurr, 2017-10-27:
ntrteam/flashcart_core#33

closed 2017-10-27 by d3m3vilurr

RisingSoul526, 2018-03-20:
Try this karnets [http://www.r4igold.cc/3ds_download.asp](url)

Omiclous, 2023-05-05:
hola, tengo la misma r4, luego de 5 años me dispuse a arreglarla, cómo lo conseguiste?