ntrteam/ds_ntrboot_flasher #35: r4sdhc stup failed

Bagou450, 2017-10-23:
r4sdhc stup failed 
i have a dual core r4 sdhc card 2017 i have followed the tutorial of 3ds.guide and https://www.youtube.com/watch?v=9lD_x2semvQ&t=304s but each time i choose my card so r4 sdhc and it's my setup failed
or I buy the r4 :  http://www.r4wood.com/products/R4-SDHC-Dual-Core-for-3DS-XL-LL-3DS-2DS-DSi.html 
the firmware inclued
my version of 3ds : new 3ds 11.5.0-38E

yavier2327, 2017-10-23:
i have exactly the same issue, i have the version 11.6 :/

jason0597, 2017-10-23:
The flasher will **not work** with r4isdhc.hk or r4isdhc.com.cn flashcarts. It will **only** work with r4isdhc.com or r4i-sdhc.com flashcarts.

rswsc0407, 2017-10-23:
I've done the exact same thing and I can't seem to get it to work either. My card says www.r4i-sdhc.com too.

yavier2327, 2017-10-24:
in this thread R4i 3ds RTS (r4i-sdhc.com) bug
#23 opened 9 days ago by kuwoh  there is the solution

Bagou450, 2017-10-24:
can you update it to make it compatible?

Bagou450, 2017-10-27:
she walks? http://www.nds-card.com/ProShow.asp?ProID=490
elle marche ellle

d3m3vilurr, 2017-10-27:
if still not work, reopen issue

closed 2017-10-27 by d3m3vilurr