ntrteam/ds_ntrboot_flasher #41: Unable to restore (remove ntrboot flash from) my Acekard 2i HW81

Machow8, 2017-10-31:
Hello,

I could flash ntrboot hax in my Acekard 2i hw81 via DS Lite, but, 3DS' ntrboot flasher program says that flashcart isn't supported.

On log file there is:

[AK2i]: HW Revision = 34b8b7ff
[AK2i]: Shutdown

Is there a way to solve this in a new version of your program?

Thanks.

Antviau, 2020-01-08:
Hi, use decript9wip it's the only thing that works in those kind of cases for some reason.

Antviau, 2020-01-08:
And I now that this is 3 years old, but someone had to answer it, I guess.

Machow8, 2020-01-09:
Thanks! No matter how long it takes for an answer like yours come, it's always welcome to find alternatives like that! =]

Antviau, 2020-01-09:
No, problem, I had the same issue and sadly had to figure that one on my
own, that really is a weird issue though.

Exayl342, 2022-06-28:
I have the same problem but i use r4i rts 3ds card