ntrteam/ds_ntrboot_flasher #43: r4 sdhc dual core 2017 error

agen4er, 2017-11-10:
When i press retail a text appears saying writing blowfish (1) 0% then another text flashes which is unreadable and then the bottom screen goes completely black without making any progress

jason0597, 2017-11-18:
Is this flashcart `r4isdhc.com`? Or is it `r4isdhc.hk` / `r4isdhc.com.cn` ?

agen4er, 2017-11-19:
.com .cn

jason0597, 2017-11-19:
.com.cn flashcarts are unsupported for ntrboot.
Please close this issue, and refer to https://github.com/ntrteam/flashcart_core/issues/33

panos4, 2017-11-20:
r4isdhc.com.cn never supported for ntrboot????

panos4, 2017-11-20:
i buy r4isdhc.com.cn and i have the same problem....what i can do???

jason0597, 2017-11-20:
@panos4 You can wait for it to be supported.

panos4, 2017-11-20:
With an update???

jason0597, 2017-11-20:
Are you Greek?

Εάν εισαι, τοτε εχω να σου πω πως πρεπει να περιμενεις για καινουργιο update για αυτα τα εργαλεία για hacking

panos4, 2017-11-20:
Ναι Έλληνας ειμαι!
Που μπορώ να ενημερώνομαι για τα καινούργια update? Μολις σήμερα αγόρασα
αυτην την flashcard και παιδευομαι όλη μέρα....

jason0597, 2017-11-20:
Οπως σου εχω πει, αυτη η r4 δεν υποστηρίζεται απο το ntrboot, θα πρεπει να περιμένεις να υποστηριχτεί

Sorry for this d3m3

d3m3vilurr, 2017-11-21:
yeah please use eng.

jason0597, 2017-11-21:
@d3m3vilurr but yeah, you can close this issue as it is a flashcart_core issue

closed 2017-11-21 by agen4er

panos4, 2017-11-21:
What this mean???