ntrteam/ds_ntrboot_flasher #44: R4i SDHC Dual Core 2016 - "Flashcart setup failed" after 3 tris

123321mario, 2017-11-18:


jason0597, 2017-11-18:
Is this flashcart `r4isdhc.com`? Or is it `r4isdhc.hk` / `r4isdhc.com.cn` ?

jason0597, 2017-11-25:
@123321mario

123321mario, 2017-12-10:
Oh crap sorry i didn't see your message, unfortunately i lost the label but, the cart circuits are not exactly the same as the official ones. After some research i found  people having the exact same cart and it's labelled www.r4isdhc.com.cn .

jason0597, 2017-12-10:
@123321mario try dumping the flash with [this](https://cdn.discordapp.com/attachments/356378901250834434/389425509307645952/ntrboot_flasher_nds.nds) flasher, by choosing the ace3ds plus option

make sure to set the log level before choosing ace3dsplus

123321mario, 2017-12-11:
It created me a folder and there's the content : 
[ntrboot.zip](https://github.com/ntrteam/ds_ntrboot_flasher/files/1548814/ntrboot.zip)

jason0597, 2017-12-11:
@123321mario [Download boot9strap_ntr](https://github.com/SciresM/boot9strap/releases/download/1.3/boot9strap-1.3-ntr.zip), and from that zip, take the `.firm` file and put it in the `ntrboot` folder that's on your flashcart's SD card.

Afterwards, boot into the flasher again, set the log level to debug, select ace3ds, and then select "Inject FIRM"

Once it finishes and kicks you back to the main menu, shut down the console, and then follow [this guide](https://3ds.guide/installing-boot9strap-(ntrboot)]

123321mario, 2017-12-11:
Thanks a lot ! It worked ! But now I Want to recover my card but the flasher don't recognize my flashcard i tried all options nothing works

jason0597, 2017-12-11:
You're following the section for restoring the flashcart that's on the guide, right?
If so, what flash dump were you planning on using? The one you uploaded earlier?

123321mario, 2017-12-11:
Yes the one i've uploaded.

jason0597, 2017-12-11:
Try using this flasher instead to restore it:
https://www.dropbox.com/s/fz187pirzqgkj3x/ntrboot_flasher.firm?dl=0

123321mario, 2017-12-11:
I will try it tomorrow, it's late :)

123321mario, 2017-12-12:
It's working ! Thanks you for all !
Did you have a discord server to recommend me if I have others problems?
I'm on the Nintendo Homebrew one

jason0597, 2017-12-12:
@123321mario I can recommend you gbatemp (not a discord server), they can help you if you ever have an issue

Apart from that, the issue has been resolved now, so, I suggest closing it

123321mario, 2017-12-12:
ok i will check it out, i'm closing the issue ;)

closed 2017-12-12 by 123321mario