ntrteam/ds_ntrboot_flasher #46: 4.0 does not boot into dsntrbootflasher, older versions do

jaydent1349, 2017-11-24:
Using the magnet + start/select/x/power button combination with R4i gold does not work with 4.0. You get a black screen and once you remove the magnet, the DS has booted normally, except the flashcard icon is empty, as it isn't detected.

I tried using version 2.2, and that version booted fine. I'm not sure about other versions before 4.0, but somewhere along the line you've made a change that has stopped something from working. I would advise looking into this, as R4i gold is the flashcard most people will be using, and so you're going to see a lot of people encountering this issue (as others already have mentioned already).

jason0597, 2017-11-25:
What R4iGold revision is this?

jaydent1349, 2017-11-26:
http://www.nds-card.com/ProShow.asp?ProID=149

panos4, 2017-11-26:
http://www.nds-card.com/ProShow.asp?ProID=507

jason0597, 2017-12-12:
@jaydent1349 If you use [this](https://cdn.discordapp.com/attachments/356378901250834434/389425509307645952/ntrboot_flasher_nds.nds) flasher, do you still get the same issue?
Note: You need to put `boot9strap_ntr.firm` (download from [here](https://github.com/SciresM/boot9strap/releases/download/1.3/boot9strap-1.3-ntr.zip)) in the `ntrboot` folder on your flashcart's SD card (make one if you don't have one).