ntrteam/ds_ntrboot_flasher #47: R4 SDHC Dual Core smart update 2017

panos4, 2017-11-25:
will be support with an update to ntrboothax soon?????
(www.r4sdhc.com.cn)

jason0597, 2017-11-25:
This is not the place to be asking for this. If you want to keep up with the latest updates on whether this flashcart will get supported, refer to https://github.com/ntrteam/flashcart_core/issues/33

Translation for you @panos4 is at: https://hastebin.com/ujivixiqor.erl

jason0597, 2017-11-25:
@d3m3vilurr Can this issue be closed?

closed 2017-11-26 by panos4

jason0597, 2017-12-17:
@panos4 do you have a gbatemp account or a discord account?

panos4, 2017-12-17:
No but i can try to create one....

Στις 17 Δεκ 2017 5:31 ΜΜ, ο χρήστης "jason0597" <notifications@github.com>
έγραψε:

@panos4 <https://github.com/panos4> do you have a gbatemp account or a
discord account?

jason0597, 2017-12-17:
@panos4 https://discordapp.com/register?redirect_to=%2F make an account, and then tell me your name
it will be on the bottom left corner, and it will have a # with 4 numbers after it

panos4, 2017-12-17:
> register?redirect_to=%2F make an account, and then tell me your name

panos444#1047

zacchi4k, 2017-12-17:
@jason0597  actually, if you have a GBATemp account, can you pm me? I have some questions about the new nds flasher

panos4, 2017-12-17:
Οκ

jason0597, 2017-12-17:
PM MrJason005