ntrteam/ds_ntrboot_flasher #50: R4 SDHC Dual Core 2014 not supported?

flipnote, 2017-11-30:
i tried using the latest flasher but it goes blank after the "writing blowfish" text what seems to be the problem?

sensei221, 2017-12-01:
having the same issue with cn card. seems like only the ones from .com site work (e.g. not .com/cn or .com/hk). Hoping there is a new release soon

zacchi4k, 2017-12-02:
I'm having the same issue too

@sensei221 
what site are you referring to?

jason0597, 2017-12-10:
@sensei221 `.hk` and `.com.cn` flashcarts are unsupported.
However, you may get your `.com.cn` working with [this](https://github.com/ntrteam/flashcart_core/issues/33)

jason0597, 2017-12-17:
@sensei221 if you are interested in getting your .com.cn working, i may be able to help you

panos4, 2017-12-17:
Ok! Thanks!!!!!!😊

HendrikGrobler, 2018-03-12:
@panos4 @jason0597 I have the same card with the same problem, please share your solution with me.

Thank you!