ntrteam/ds_ntrboot_flasher #51: Doesn't work using a brand new official R4i GOLD 3DS RTS (r4ids.cn)

t00bib, 2017-12-06:
After the flashing blank screen, I got a black screen (on both screens)... :((

t00bib, 2017-12-06:
Flashcart bought from http://www.nds-card.com and received today.
WOOD R4 V1.64 Released. (2013-11-08) installed.
Brand new Nintendo New 2DS XL v.11.5.0-38E (just bought, never used)

flipnote, 2017-12-07:
probably it wont work on .cn on .com only mine has an issue too its blank also its from .hk

t00bib, 2017-12-07:
Problem solved ^^
It seems it didn't work on a freshly formated, empty SD card.

Here is the workaround:
  1. Download the latest version of the WOOD R4 Kernel (http://filetrip.net/dl?FhXOBrY7vI).
  2. Extract and copy all files of the archive on the root of your flashcart SD card.
  3. Then simply add ds_ntrboot_flasher_dsi to the root of the SD card.

Et voilà! :)

PS: If the download link doesn't work, check on http://www.r4ids.cn/r4i-download-e.htm to find a mirror link.

closed 2017-12-07 by t00bib